KivApple/dell-charger-emulator #1: No Hardware Instructions

alxpettit, 2017-10-23:
It would be useful to see the wiring schematic you used for this project. Perhaps you could add it to README.md?

KivApple, 2017-10-24:
I have wrote in README.md:

> You need ATTINY25, ATTINY45 or ATTINY85 MCU (you may need to change compiler flags in Makefile) running at 8 MHz (the internal RC-oscillator is suitable, but if desired, you can use an external crystal - adjust the fuses accordingly). The Dell power adapter connector has three pins - GND, VOUT (19V) and ID. Connect the ID to the PB2 pin of the microcontroller. Also provide 3.3V power for MCU (you can use a simple linear regulator 78L33 to make the desired voltage from 19V).

The connection scheme is present in the board directory in KiCad format, there is also a proposed printed circuit board. But since you ask, I added a picture to README.md.

The Python script is able to flash EEPROM through the 1wire only of the emulator on the AVR, it is not able to flash the real DS2502, because, first, it has an add-only memory, and secondly, it needs to supply the increased voltage at the right moment for programming. However, a script in Python is able to read the contents of both the emulator and the real memory chip. Of course, for the emulator to respond to the 1wire protocol, you need to download my firmware into it via any AVR programmer (for example, USBASP).

Perhaps this scheme will work on "large" microcontrollers like ATMEGA328 (used in many Arduino). In principle, all that the program needs is one external interrupt (INT0 usually) and one timer with one comparison channel. And a couple of kilobytes of ROM. However, AVR microcontrollers have some differences and most likely you will need a small edit of the sources to make everything work.

I used ATTINY, because in this way I could make a PCB of the smallest size and fit it into the plug of the power adapter. In addition, the ATTINY85 chip is cheaper than the ATMEGA328 chip.

alxpettit, 2017-10-28:
I've gotten arduino nano clones for as little as $3 per unit -- they are surprisingly cheap. I agree that they are definitely overkill for this project, and certainly wouldn't give you a great form factor, but general purpose programmable chips are overkill for a lot of their applications. I'd argue that the point of a programmable chip is to do lots of possible functions, some simple, and some really complex.

It only makes economical sense to make programmable chips that do simple things, because the less _unique_ things industry has to make, the cheaper it is to make them -- we can optimize via creating generic solutions and retooling them to every possible use case.

It's also good as a way to test an idea before you invest in more specialized hardware, I think. But Tiny85s are so cheap on Ebay, that I bought a couple of them just for testing on this project. I just think other people might not be willing to spend money experimentally, though -- and it's a shame, if they're barred from trying this project out because they don't have the hardware.

In my case, I can't really imagine a use for Tiny85s besides weird little things like this -- or just getting an LED to blink with a specific pattern, or doing some very very simple logic control... usually when I'm working with hardware, I'm working with fairly advanced robotics. (And furthermore, I'm normally a high level software gal. System administration, Python, that sort of thing. I rarely end up doing manual memory management, and the most low level thing I do is C++ coding on Arduinos, usually.)

So firmware-level things like this are out of my usual scope. I freely admit I could be talking out of my excretion hole here. And I totally respect if you don't want to put the effort in to translate it over to an arduino. :P

Oh, and sorry I didn't recognize the KiCad. I've never used it in my life before. The closest I've ever come is using a perfboard. Thank you for adding the PNG schematic!

closed 2017-10-28 by alxpettit

TehseenHasan, 2020-06-27:
I made a complete tutorial: https://github.com/KivApple/dell-charger-emulator/issues/4#issuecomment-650623879

KivApple, 2020-06-27:
@alxpettit If anyone wants to add support for different AVR MCUs (such as ATMEGA328 which is used in Arduino Uno) feel free to create a pull request. You can easily upload plain C code into Arduino board. It is not required to use Arduino libraries when you are programming Arduino so you can write code which will be compatible both with ATTINY85 and ATMEGA328 (with few #ifdef's).

lgolebio, 2022-04-16:
Hi, I'm totaly noob at electronics and arduino. I have one question. Why it is not possbile to upload code to Digispark Attiny85 through built in USB port ?