KivApple/dell-charger-emulator #2: eeprom-data.hex is a bit confusing to edit

alxpettit, 2018-01-31:
So, `eeprom-data.hex` isn't a binary file. I was originally confused about this -- trying to open it in a hex editor and such, before I realized it was "meta-hex" -- ASCII values encoded in hexadecimal.

This seems confusing to edit, and none of my coding tools will actually interpret the hex data into ASCII in any panels, so I can see what it says easily. E.g., where the "DELL" tag is, etc.

Now, of course, I can do all this by hand, but it doesn't seem ideal. I'm weary of making a mistake, since I could theoretically fry a motherboard that way.

Is this ASCII-encoded *.hex file a standard format? If so, what tools can I use to edit it? (PLEASE let them be FOSS tools! :stuck_out_tongue_closed_eyes:)

If _not_, would it be worth considering having this be a normal binary file instead of ASCII-encoded?

KivApple, 2018-02-02:
Intel hex is standard format for firmware, which recognized by avrdude.
You can use my python script to change changer data.

alxpettit, 2018-02-02:
@KivApple Ah Is there a standard tool for editing it that you know of, with features like a hex editor? Or even better a hex editor that will also except intel .hex files? I like to know how to easily do things by hand, in case I need to. :stuck_out_tongue:

closed 2018-02-02 by alxpettit