KivApple/dell-charger-emulator #3: Possibility of using "arduino as ISP" to program attiny85 instead of 1wire?

alxpettit, 2018-07-28:
My current preferred setup for attiny programming consists of a board with an arduino programmed to function as a ISP, with a couple of additional components to prevent it from resetting to bootloader. The relevant wiring is essentially [this](https://create.arduino.cc/projecthub/arjun/programming-attiny85-with-arduino-uno-afb829?f=1). This is a very popular type of programmer I think, since a lot of people have and know where to get arduinos.

How would I use this project with my setup to program my attiny85?

hetii, 2019-01-02:
You can just compile this project by using its makefile and grab hex file to use with your arduino like programmer.
You can also change makefile and edit programmer type to match your programmer.

alxpettit, 2019-04-03:
Wow, this has been sitting here a long time!

For anyone curious, here's what I did to push the hex via my generic atmega328p board:
`avrdude -c arduino -P /dev/ttyUSB0 -p attiny85 -U flash:w:dell-charger-emulator.hex -b 19200`

It took a bit of poking around for me to figure out how to do this. Eventually I realized the easiest thing to do was simply try flashing via `Arduino as ISP` and see what options were sent. Turns out `-b 19200` was a prereq.

For anyone unfamiliar with `avrdude`:
`-c arduino` sets the programmer to `Arduino as ISP`, `-P` sets the port to the port of the arduino, and `-p` sets the AVR chip to compile for. `-U` is for operation.

closed 2019-04-03 by alxpettit