KivApple/dell-charger-emulator #4: Can I use this code in Arduino Uno?

TehseenHasan, 2020-06-26:
Can I use this code and upload it into arduino? I have also same problem of the Dell charger.
and another question is that, if I use ATTINY85 then How Can I upload this complete code into ATTINY85?

TehseenHasan, 2020-06-27:
I figured it out. I used Arduino Uno as ISP Programmer to program the Digispark attiny85 module. Then I successfully uploaded the `dell-charger-emulator.hex` and `eeprom-data.hex` files into Attiny85 microcontroller by using the `avrduid` commands.

**This is the step by step tutorial I made for anyone who is doing the same thing:**

### Installing Arduino IDE and Attiny Microcontroller Boards:
Install the Arduino IDE. [Link to Download](https://www.arduino.cc/en/main/software)
Open the Arduino IDE.
By default Arduino IDE doesn't support ATtiny85 so we should add ATtiny boards to Arduino IDE. 
Open File -> Preferences and in the Additional Boards Manager URLs and paste this url there: `https://raw.githubusercontent.com/damellis/attiny/ide-1.6.x-boards-manager/package_damellis_attiny_index.json`

[image]

After this is done open Tools -> Board -> Board Manager

[image]

After opening Board Manager scroll down the list where it says "attiny by Davis A. Mellis". Click on that and install it.

[image]

After installing now you would be able to see a new entry in the Board menu

[image]

### Connecting ATtiny85 with Arduino Uno
Now with all the above things ready we will start programming the attiny85. Connect the arduino uno to attiny85 using jumper wires as below:

[image]

Additional Hint (Optional): Add a 10uF capacitor between RESET and GND in arduino. This is to avoid arduino from being auto reset when we upload the program to attiny85. If you are using a electrolytic capacitor make sure the anode goes in GND of uno.

### Uploading program to ATtiny85
Now back to Arduino IDE. Select ATtiny under Tools -> Board. Then select ATtiny85 under Tools -> Processor. And select 8 MHz (internal) under Tools -> Clock.

[image]

[image]

Then make sure Arduino as ISP is selected under Tools -> Programmer

[image]

By default the ATtiny85 runs at 1MHz. To make it to run at 8MHz select Tools -> Burn Bootloader. 

[image]

[image]

You will get the above message if burning bootloader was successful.

Now for testing purpose, open the Blink example from arduino examples and change the pin number from 13 to 0 and upload.

[image]

[image]

You can see the above message if everything was successful. Now we have upload the blink program to ATtiny85.
Now lets upload the actual dell-charger-emulator.hex` and `eeprom-data.hex` HEX files into the ATtiny85.

### Uploading dell-charger-emulator.hex` and `eeprom-data.hex` HEX in to ATtiny85

TehseenHasan, 2020-06-27:
a great benefit of using Digispark ATtiny85 Board is that it is **on-board voltage regulator**. So we can provide the 19.5 voltes of Dell Charger directly to the ATtiny85 Board on its `VCC pin`. So no need of external Voltage regulator on Digispark ATtiny85 Board.

tylerwight, 2022-04-20:
@TehseenHasan I actually got the same digispark attiny85s before coming here and realizing you did also. I was able to follow your directions and flash both hex files just fine. Thanks for your help! I also used avrdude to read the eeprom and it looks like the data I put there. Your post was very helpful and I appreciate it.

I am trying to get this working on 2 dell laptops that I got for free with no power adapter at all. Did the default hex work for you? The default specs for me are correct (45w, 19.5v, 2.3a) for me so I left them that way. Did you have to change your eeprom hex at all to get it working?

the laptops work fine they just don't charge the battery. The charger header has power ground and a single wire (I assume it's ID). They are older inspirons with i3 6xxx processors so I assume they are old enough the methods should still apply.

tylerwight, 2022-04-21:
I'm coming back here to provide an update to my last post:

The laptops I was testing with before were the exact same model, and they were 2-in-1s that flipped over, very tiny charge port that is not like the common dell barrel plug.

I did happen to have a different dell laptop that had the more standard barrel plug, and I also had an HP branded power adapter that fit exactly perfect including the center pin for data. So in order to test I took my programmed attiny85 as seen above and put it into the HP charger along the ID wire, and it works!

So it seems the dell Inspiron 11 3153's use a different type of charger that reports differently, so the code in the repository will not work for them. This is interesting though because all 3 laptops, the one the works and the 2 that don't, are all intel 6th gen processors, so they should be pretty close in age.

lgolebio, 2022-04-23:
I can confirm that on some Dell laptops communication with attiny85 doesn't work. It still shows unknown ac adapter. Maybe it is due to clock of attiny85 ?

lgolebio, 2022-04-24:
Hi. Just to inform anyone. I found a solution.

1.  Go to "tools => Library Manager" 
2. In Filter search type "onewirehub" and install version 2.2.1
3. Go to "Examples => OnewWireHub => DS2502_DELLCHG"
4. Modify "**constexpr uint8_t pin_onewire   { 9 };**" to  "**constexpr uint8_t pin_onewire   { 2 };** " 
5. The pin P2 of digispark is our onewire bus.
6. Upload the code to digispark
7. Works on all Dell laptops I have !

dainiusb, 2022-07-19:
I tried the one by sekyHC, this one and DS2502_DELLCHG example. Nothing seems to work with Dell Latitude D620.

dainiusb, 2022-07-20:
I'm measuring around 75 mV on the pin when chip isn't connected and on my oscilloscope I'm seeing low pulses of around 500 microseconds, is this normal?

dainiusb, 2022-07-20:
Nope, found a bad transistor.

dainiusb, 2022-07-21:
After repairing the laptop I tried the DS2502_DELLCHG example and it works!

FBMinis, 2023-01-22:
> Can I use this code and upload it into arduino? I have also same problem of the Dell charger. and another question is that, if I use Digispark ATTINY85 then How Can I upload this complete code into Digispark ATTINY85? [image]
> 
> I am using this Digispark attiny85 module and It can be programmed using Arduino IDE. Please help me that how can I program your code in this attiny85 module using Arduino IDE?

I followed your instructions and it worked perfectly.

I used the blue Digistump and in the Blink sketch I had to change LEDBuiltIN to 1.

When uploading the ArduinoISP sketch to the Uno, I used `#define USE_OLD_STYLE_WIRING` and used pins 10,11,12 and 13
[image]

vusalyusifov-diacom, 2026-03-02:
I know this method. I already have ready-made DS2501 chips. It's possible to arrange this with just coding, just one chip and a special PCB. I have a project related to this. Let's see how it turns out.