KivApple/dell-charger-emulator #5: UART Schematic: Pullup needed?

mihaigalos, 2022-02-24:
Hi, not sure the schematic in its current form would work:
```
TXD --------|<|----*-------- ID
                   |
RXD ----------------

GND ------------------------ GND
```
Seems `TXD` is at least missing a pullup on the anode (assuming the diode polarity is correct), since when you want to send, you would drive the pin low and open the diode:

```
                        VCC 
                        _|_
                       |  |
                       |__|
                        |
TXD --------|<|----*----*--- ID
                   |
RXD ----------------

GND ------------------------ GND
```

lgolebio, 2022-04-21:
Yes. I can confirm that pullup resistor is needed 4,7Kohm. Without it it reads data but incorrect. 

[image]

rahmajayagrs, 2022-09-16:
read correct , but write fail
[image]

mihaigalos, 2022-09-16:
@rahmajayagrs, so the conclusion is that it only works without the pullup?

rahmajayagrs, 2022-09-16:
with or without pullUp still can read but can't write

mihaigalos, 2022-09-19:
@rahmajayagrs the value of the pullup is important, as is the maximum baud rate.
Try using a lower baudrate (i.e. `9600`) and a resistor value of `4.7K`.

Also, double check your wiring.

arekm, 2024-07-10:
Here I was able to read EEPROM contents of original 130W Dell charger and also of emulated EEPROM but only with 4.7k pullup. No speed change was needed. My USB adapter had FTDI chip in it.

Didn't try to reprogram emulated EEPROM that way since Dell XPS 9530 I was testing recognized emulated charger properly. Programming was done in place on digispark attiny85 board, using TL866II Plus with SOIC8 clip. Fuse bits set the same way as in project Makefile.